icl-rocketry/art-sirius #1: Ignitor Isolation Board

shreybohra, 2024-03-02:


shreybohra, 2024-03-02:
Board Requirements:

- 2x power inputs from power distribution boards
    - Power available indicator LEDs for each
- 2x 3.5mm jacks to act as flight pins
    - Indicator LED to show safe/unsafe condition
- 8x signal inputs
    - 4x from Blue Raven, 4x from Pickle
- 4x signal outputs
    - BP A, BP B, Motor A, Motor B
- MOSFETs to switch firing on/off

SimeonCharles, 2024-03-20:
@shreybohra , initial pcb design, i think the voltage and ground traces are too thin, what thickness do you recommend for those? Also I think there is a nicer way of laying out the signals, I'll try another way tomorrow

[image]

shreybohra, 2024-03-20:
looking good! I'll review it properly after the weekend. Are the .sch and .brd files on GitHub now?

re the traces - the current going through the traces shouldn't last for more than a second so trace width isn't as much of a concern as it would normally be for something like this, but I agree that they look a bit thin. 

use a PCB trace width calculator online to see what they suggest - remember that those calculators give you the recommended trace width for a continuous load, and they normally have a lower temperature cap then what we're comfortable with, so if the trace width they suggest is unreasonably high then don't worry about it.

shreybohra, 2024-03-29:
@SimeonCharles wondering if you've updated this at all! let me know when you want me to review again :)

closed 2024-04-28 by shreybohra